at-wat/neonavigation #31: some issue in trajectory_tracker.cpp

ipatient-zju, 2017-05-15:
Hi, I am a student of Zhejiang University, and I am interested in your neonavigation stack. When I focus on the trajectory_tracker.cpp, I can not understand some codes as follows:
   First,how to understand this formula?
origin.x = cos(w * lookForward / 2.0) * v * lookForward;
origin.y = sin(w * lookForward / 2.0) * v * lookForward;   
   Second, I can not understand the timeoptimal_control function, is there some papers to explain it? so in the control part, the velocity calculate function is v = timeoptimal_control(-remainLocal * signVel, acc[0], dt); what does this mean?
   Third, the path used in the trajectory_tracker is a local plan or a global plan?  
   Hope for your reply, thanks!

yonhdee, 2017-05-17:
#3 the input path is any path (nav_msgs/Path) the tracker node subscribe to, with a little constraints

at-wat, 2017-05-21:
The formula of origin.x and y is the motion prediction using linear extrapolation.

Time optimal control in this node is a position control under given maximum velocity and acceleration.
You can easily derive the formula of the velocity in this function from:
```
x(t0) = 0
d/dt x(t0) = 0
(d/dt)^2 x = constant
```

The input of the node can be any path. Note that the path should not be too sparse since this node calculates curvature from the poses in the path.

at-wat, 2017-07-05:
Closed.
If you have any more question, please reopen or open a new issue.

closed 2017-07-05 by at-wat