at-wat/neonavigation #32: tie_maps error

yoichims, 2017-05-31:
Hi, I am using ros indigo and having the following error:

tie_maps.cpp:44:3: error: ‘MapMode’ was not declared in this scope
   MapMode mode;

I brought map_server from repository to compile in the catkin_ws, however, it did not solve the problem.

I am using ROS indigo.

My question is if neonavigation only works with kinetic?

Thank you.

at-wat, 2017-06-01:
tie_maps depends on a library in map_server and ``MapMode`` in it seems not available on Indigo.

The easiest fix is cloning latest map_server in the workspace in which neonavigation is placed.
It is better to put neonavigation and related things in separated overlay workspace.

yoichims, 2017-06-01:
Thank you.
I tried but I had some other problems with 
undefined reference to 
 In function `tie_maps_node::tie_maps_node()': 
`map_server::loadMapFromFile(nav_msgs::GetMapResponse_<std::allocator<void> >*, char const*, double, bool, double, double, double*, MapMode)'

I think this can be solved by modifying source a little bit.
I will take a look later next week and update.

at-wat, 2018-07-07:
Now, this package is able to be built on Indigo. (https://github.com/at-wat/neonavigation/pull/180)

closed 2018-07-07 by at-wat