at-wat/neonavigation #76: Search timed out

yutakaoka, 2017-09-28:
Hi,
I just tried it with neonavigation_launch/navigate.launch on Fetch robot, I am sure remapping or map loading seems fine, but "Search timed out" warning came out a lot, and ended up stop and go motion.
I am using indigo.
Is there any way to make it smoother? 
Any info would be appreciated.
Thanks,

at-wat, 2017-09-28:
Hi,
The performance of the planner\_3d node pretty much depends on the optimization level.
A 3-DOF gridmap class in this node uses a kind of CPU cache optimization technique.
Release-build should help. (e.g. `$ catkin_make --cmake-args -DCMAKE_BUILD_TYPE=Release`)

yutakaoka, 2017-09-28:
Cool. It worked. Thanks!

yutakaoka, 2017-10-02:
When robot is approaching where it is close to the goal position, it is easy to pop up "Search timed out" warning. It seems like if there is a big difference between approaching angle and goal pose, it happens a lot. Any idea to figure out this situation?

at-wat, 2017-10-02:
> https://github.com/at-wat/neonavigation/issues/76#issuecomment-333448705

This is caused by searching detailed trajectory to reach final pose.
The default parameters try to avoid in-place turn since the node was originally developed for a tracked vehicle; this constraint might increase searching cost pretty much.

Decreasing `cost_in_place_turn` parameter might reduce the computation cost.
https://github.com/at-wat/neonavigation/blob/master/planner_cspace/src/planner_3d.cpp#L1077

yutakaoka, 2017-10-03:
I tried that, and actually it tended to have a in-place turn, but I saw the error on the middle of the way instead. Finally I noticed the default value(30.0) is the best for now. Is there any other parameter to make it better?

at-wat, 2017-10-05:
`Search timed out` warning is shown if the path search takes more than `1 / freq_min` seconds. After this warning, it continues searching without physical motion.
If the speed of the robot is slow enough, decreasing `freq_min` reduces the warning.
Default value: 2.0 Hz https://github.com/at-wat/neonavigation/blob/2db8a6e84a20215045a991652f9a923575e9f275/planner_cspace/src/planner_3d.cpp#L1066-L1067

`local_range` parameter is the range of the full 3-DOF searching.
Decreasing this value reduces computation cost.
Default value: 2.5 m https://github.com/at-wat/neonavigation/blob/2db8a6e84a20215045a991652f9a923575e9f275/planner_cspace/src/planner_3d.cpp#L1093

Turning `find_best` parameter false makes path search algorithm depth-first instead of optimal search. I'm not sure it works well.
https://github.com/at-wat/neonavigation/blob/2db8a6e84a20215045a991652f9a923575e9f275/planner_cspace/src/planner_3d.cpp#L1098

BTW, can you post information about your CPU and memory?
The default parameters were originally turned for (maybe 5th gen.) Core i7.

at-wat, 2018-03-19:
@yutakaoka I'm adding multi-threading feature to the planner_3d node. https://github.com/at-wat/neonavigation/pull/118
If you have remaining CPU power, increasing `num_threads (int, default: 1)` parameter might suppress `Search timed out`.

at-wat, 2019-12-28:
Some optimizations were added recently and the performance should be improved. (#315, #342)
Feel free to reopen if any related problem.

closed 2019-12-28 by at-wat