at-wat/neonavigation #77: Two sensor inputs for map_overlay

yutakaoka, 2017-09-28:
Hi,

Is there any way to deal with two sensor inputs(e.g laser and pointcloud) for making a "map_overlay" at once? I just wanted to utilize laser and point cloud info as one obstacle map.

Thanks,

at-wat, 2017-09-30:
It is not yet supported by the nodes in the costmap_cspace package.
The simplest workaround is using `laser_filters scan_to_cloud_filter_chain` without any filter plugin to convert LaserScan to PointCloud2.
`pointcloud2_to_map` node supports multiple sources of the input cloud on one `cloud` topic.

yutakaoka, 2017-10-02:
Thank you for your advice. It seemed to be working well.

at-wat, 2018-03-29:
https://github.com/at-wat/neonavigation/pull/106 added the feature to overlay multiple costmaps.

closed 2018-03-29 by at-wat