at-wat/neonavigation #81: Omni drive base

yutakaoka, 2017-11-02:
Hi,

Is there any function of `trajectory_tracker` for omni-base?
Does it support for diff-drive only for now?

Thanks,

at-wat, 2017-11-03:
Currently, `trajectory_tracker` node only supports diff-drive control.
The node is based on the diff-drive specific control method (http://ieeexplore.ieee.org/document/174452/).

HappySamuel, 2019-11-14:
Hi @at-wat 

Does the current version of trajectory_tracker support the skid steer drive system?

Best,
Samuel

at-wat, 2019-12-28:
Skid steer drive can be approximated as a diff drive with a limitation of minimum turning radius.
`planner_3d` has a parameter `min_curve_raduis` and specifying this may generate a path suitable for skid steer.

HappySamuel, 2020-01-02:
Hi @at-wat 

For this `min_curve_raduis`, for skid-steer drive shall use value lesser than 0.1 ? or larger?

Best,
Samuel