at-wat/neonavigation #93: Test failed

yutakaoka, 2017-12-05:
I noticed one of tests was failed. Did you notice that?
```
10:39:12 testtest_navigate ... ERROR!
10:39:12 ERROR: max time [120.0s] allotted for test [test_navigate] of type [planner_cspace/test_navigate]
10:39:12   File "/usr/lib/python2.7/unittest/case.py", line 331, in run
10:39:12     testMethod()
10:39:12   File "/opt/ros/indigo/lib/python2.7/dist-packages/rostest/runner.py", line 148, in fn
10:39:12     self.test_parent.run_test(test)
10:39:12   File "/opt/ros/indigo/lib/python2.7/dist-packages/rostest/rostest_parent.py", line 132, in run_test
10:39:12     return self.runner.run_test(test)
10:39:12   File "/opt/ros/indigo/lib/python2.7/dist-packages/roslaunch/launch.py", line 681, in run_test
10:39:12     (test.time_limit, test.test_name, test.package, test.type))
10:39:12 --------------------------------------------------------------------------------
10:39:12 
10:39:12 [ROSTEST]-----------------------------------------------------------------------
10:39:12 
10:39:12 
10:39:12 SUMMARY
10:39:12  * RESULT: FAIL
10:39:12  * TESTS: 0
10:39:12  * ERRORS: 1
10:39:12  * FAILURES: 0
```

at-wat, 2017-12-06:
This test performs navigation in a simulated environment.
`ERROR: max time [120.0s] allotted for test` means time-out of the test.
It can occur on machines without enough CPU power or on builds lacking optimization option.

Currently, the test is automated on Travis-CI.
The Same failure sometimes occurs when Travis is overloaded.
It might be better to increase max time in test launch file.

yutakaoka, 2017-12-08:
I got it. I will tweak the timeout parameter.
Thanks.

PR #98 (2018-01-18): planner_cspace: increase navigation test time limit
fixes #93
closed 2018-01-18 by at-wat