at-wat/neonavigation #110: Coud you please provide some bag file for testing this great package

weisongwen, 2018-01-29:
Dear @at-wat  

Thanks for your effort on this great package, Could you please provide some datesheet or bag file related to this package to test this package? 

Thanks a lot,
Best 
Welson

at-wat, 2018-02-01:
👍

brainhenry, 2018-03-11:
thanks for the help. my email address is 14108209d@connect.polyu.hk or brainhenry100@gmail.com

at-wat, 2018-04-15:
@weisongwen I have added a launch file and sample map for a demonstration.
It is not easy to test with a bag file since the navigation works with the feedback from the environment according to the motion of the vehicle. So, I have added a launch file with the kinematics (not dynamics) simulation of the vehicle and simulated local costmap.

Please try it:
```shell
$ roslaunch neonavigation_launch demo.launch
```

weisongwen, 2018-04-16:
Hi, @at-wat  
Thanks for your kind reply, I will test this first,
Best 
Welson

at-wat, 2018-05-09:
Feel free to reopen if you have any related questions or requests.

closed 2018-05-09 by at-wat

weisongwen, 2018-05-09:
Dear @at-wat 
Thanks for your kind information,

best 
Welson