at-wat/neonavigation #177: REAL ROBOT

lucabonamini, 2018-06-29:
Hi!
Is it possible to obtain a local_map for static obstacle avoidance from sensors like lidar or camera rgbd?

at-wat, 2018-07-07:
Of cause, this package is for real robots.
*laserscan_to_map* and *pointcloud2_to_map* nodes in the *costmap_cspace* package would be a good start point.
These nodes are good enough for 2-D sensor configuration in most case.
For 3-D sensors, extra obstacle detection algorithm for your robot configuration and environments might be necessary for robust motion.

lucabonamini, 2018-07-09:
Thank you for the answer! I'll try as soon as possibile! Another question, if I can: where do I have to modify the frame names in order to have the TF of the odometry and the sensors refered to the base_link of the robot?

at-wat, 2018-07-10:
It works with the frame names used in move_base.

at-wat, 2019-12-28:
Closing. Feel free to reopen, or open a new issue, if you have any question on this topic.

closed 2019-12-28 by at-wat