at-wat/neonavigation #254: safety_limiter: sometimes test_safety_limiter fails

jojo43, 2018-12-18:
Sometimes `test_safety_limiter` fails.
The test result at the time of failure is as follows.

**build/test_results/safety_limiter/rosunit-test_safety_limiter.xml**

```xml
<?xml version="1.0" encoding="UTF-8"?>
<testsuites tests="7" failures="1" disabled="0" errors="0" timestamp="2018-12-18T06:06:30" time="42.03" name="AllTests">
  <testsuite name="SafetyLimiterTest" tests="7" failures="1" disabled="0" errors="0" time="42.03">
    <testcase name="Timeouts" status="run" time="13.446" classname="SafetyLimiterTest"/>
    <testcase name="CloudBuffering" status="run" time="1.103" classname="SafetyLimiterTest">
      <failure message="/catkin_ws/src/neonavigation/safety_limiter/test/src/test_safety_limiter.cpp:205&#10;The difference between msg-&gt;linear.x and 1.0 is 1, which exceeds 1e-1, where&#10;msg-&gt;linear.x evaluates to 2,&#10;1.0 evaluates to 1, and&#10;1e-1 evaluates to 0.10000000000000001." type=""><![CDATA[/catkin_ws/src/neonavigation/safety_limiter/test/src/test_safety_limiter.cpp:205
The difference between msg->linear.x and 1.0 is 1, which exceeds 1e-1, where
msg->linear.x evaluates to 2,
1.0 evaluates to 1, and
1e-1 evaluates to 0.10000000000000001.]]></failure>
    </testcase>
    <testcase name="SafetyLimitLinear" status="run" time="3.212" classname="SafetyLimiterTest"/>
    <testcase name="SafetyLimitLinearEscape" status="run" time="2.707" classname="SafetyLimiterTest"/>
    <testcase name="SafetyLimitAngular" status="run" time="6.218" classname="SafetyLimiterTest"/>
    <testcase name="SafetyLimitAngularEscape" status="run" time="2.703" classname="SafetyLimiterTest"/>
    <testcase name="NoCollision" status="run" time="12.639" classname="SafetyLimiterTest"/>
  </testsuite>
</testsuites>
```

at-wat, 2018-12-18:
related to https://github.com/at-wat/neonavigation/pull/243/

at-wat, 2019-05-23:
#279

closed 2019-05-23 by at-wat