at-wat/neonavigation #255: move_base_msgs is the same with navigation?

gongyue666, 2018-12-24:
hi,the action msg of move_base is the same with the one navigation used?I get a running error for the action msg type not match correctly.I do not find the move_base_msgs in neonavigation_msgs, waiting for your reply.thx ahead

at-wat, 2018-12-24:
@gongyue666 Yes, planner_cspace/planner_3d uses navigation's move_base_msgs package to make it compatible with move_base.
What version of neonavigation (source build or ubuntu apt?) and which version of rosdistro do you use?

gongyue666, 2018-12-25:
hi,thx for reply. I use source build and the branch is master, my ROS version is Kinetic.the error log is blow:
[ERROR] [1545632900.403266085]: Client [/app/device_service] wants topic /move_base/result to have datatype/md5sum [move_base_msgs/MoveBaseActionResult/3ffefadf6d1af0bc65d2848165d97fae], but our version has [move_base_msgs/MoveBaseActionResult/1eb06eeff08fa7ea874431638cb52332]. Dropping connection.
[ERROR] [1545632900.428007096]: Client [/app/device_service] wants topic /move_base/feedback to have datatype/md5sum [move_base_msgs/MoveBaseActionFeedback/aae20e09065c3809e8a8e87c4c8953fd], but our version has [move_base_msgs/MoveBaseActionFeedback/7d1870ff6e0decea702b943b5af0b42e]. Dropping connection.

at-wat, 2019-01-07:
@gongyue666 could you check a result of the following command?
```shell 
$ rospack find move_base_msgs
```
This should return `/opt/ros/kinetic/share/...`.

Apt-get installed `move_base_msgs/MoveBaseActionResult` should have md5 of `1eb06eeff08fa7ea874431638cb52332`. I guess you are using different version of the `move_base_msgs` in your local catkin_ws.

```
[ERROR] [1545632900.403266085]: Client [/app/device_service] wants topic /move_base/result to have datatype/md5sum [move_base_msgs/MoveBaseActionResult/3ffefadf6d1af0bc65d2848165d97fae], but our version has [move_base_msgs/MoveBaseActionResult/1eb06eeff08fa7ea874431638cb52332]. Dropping connection.
```
means that `Client` (your app) requests `3ffefadf6d1af0bc65d2848165d97fae` which is different from apt-get installed version.

at-wat, 2019-06-22:
The problem looks caused by the client side since the message md5 hash used by planner_3d node is correct.

Please reopen if the problem still exists.

closed 2019-06-22 by at-wat