at-wat/neonavigation #325: Error: Given goal is not on the map

HappySamuel, 2019-06-14:
Hi

I tried to use mcl_3dl together with the neonavigation in Gazebo simulation. 

[image]

However, when i tried to send a goal on the map in RVIZ, the error appear as below:

[image]

And i refer back to running demo.launch. I found that the goal sent via RVIZ is 0 elevation in z-axis. Please advise how should i send a goal without elevation in z-axis. 
Thanks.

Best,
Samuel

at-wat, 2019-06-14:
Z position is simply ignored in planner_3d node.
Third value shown in `New goal received` INFO message is yaw orientation.

I guess you are sending a goal on non-map frame. (related to a bug #324 )
Please try setting `Global Options`/`Fixed Frame` of rviz to map.

HappySamuel, 2019-06-14:
Hi @at-wat 

I checked my RVIZ configuration again and found that i am having the setting as you said Global Options / Fixed Frame with map. But the problem still the same. What shall i do next?

Best,
Samuel

HappySamuel, 2019-06-17:
Hi @at-wat 

Another discover, i tested your neonavigation demo with my map (pgm or png), but the error is the same. "Given goal is not on the map". So i guess the problem lies with the map? My map size is 2048x2048. Please advise what shall i do now? 
Thank you.

Best,
Samuel

at-wat, 2019-06-17:
@HappySamuel could you attach map yaml file (or header part of `rostopic echo /map`) and `rostopic echo /move_base_simple/goal`?

HappySamuel, 2019-06-19:
Hi @at-wat 

I found that if i use a map that is created by 2D lidar. It can works well. Do you know any tools that i can use for creating the 2D map from 3D lidar pointcloud or pcd map?

Best,
Samuel

at-wat, 2019-06-19:
@HappySamuel please attach example data which doesn't work well.
Map yaml file, header part of `rostopic echo /map`, and result of `rostopic echo /move_base_simple/goal` might help.

at-wat, 2019-06-19:
> Do you know any tools that i can use for creating the 2D map from 3D lidar pointcloud or pcd map?

@HappySamuel I used to use `pointcloud_to_maps` node in neonavigation/map_organizer package. (https://github.com/at-wat/neonavigation/tree/master/map_organizer#pointcloud_to_maps)
But, it is not well maintained at now and I'm not sure it works well.

HappySamuel, 2019-06-20:
Hi @at-wat 

I accidentally deleted the failed map. Then I tried to reproduce the same error. But i couldn't. It's weird. I use the same procedure that i created map last time, which result the above issue. But the map i produced. there is no error. I am sorry that i couldn't provide you the information you need.

Best,
Samuel

HappySamuel, 2019-11-21:
Hi @at-wat 

I found the cause of the error as described from beginning. It's caused by the 3rd element of origin as non-zero value

> origin: [65.20000, -59.40000, 1.5707963]

[image]

But when i changed the 3rd element of origin to zero as shown as below:

[image]

Only then the move_base can receive the goal.

But when i tried to change the 3rd element of origin to 0.1, the result become as below.

[image]

Do you know why?

Best,
Samuel

at-wat, 2019-12-11:
> 3rd element of origin

@HappySamuel Currently, costmap_cspace package doesn't handle the 3rd element of origin (yaw orientation), but rviz looks do.
costmap_2d and amcl in navigation stack seems also ignoring orientation.

HappySamuel, 2019-12-11:
Hi @at-wat 

Is that means the 2D map i created must be having the same origin (X,Y,yaw) as 3D map i used?

Best,
Samuel

at-wat, 2019-12-11:
Yes, coordinate frame of localization and navigation must be matched.
x, y can be adjusted by using 2D map origin in yaml, but orientation should be aligned beforehand.

at-wat, 2019-12-28:
#401 will track the rotation component support for OccupancyGrid origin.
Feel free to add comments to there and/or creating PRs.

closed 2019-12-28 by at-wat