at-wat/neonavigation #329: Guidance needed on using local costmap with 3d lidar realtime.

HappySamuel, 2019-06-19:
Hi

I have running this package on Gazebo Simulation. Please advise how should i implement the 3d lidar sensing to do the obstacle avoidance. I found that the demo.launch show the dummy robot can do obstacle avoidance with a attached static map with obstacles registered. How shall real time obstacle sensing be done and registered into the map? So that the dummy robot can avoid the obstacle, which is on the path to goal.

Best,
Samuel

at-wat, 2019-06-21:
costmap_cspace/pointcloud2_to_map and costmap_cspace/laserscan_to_map nodes can be used for this purpose.

https://github.com/at-wat/neonavigation/blob/f13824534e07a8d8547cf9d97648218e1315b27a/neonavigation_launch/launch/demo.launch#L14-L19
Please try replacing costmap_cspace/largemap_to_map in the demo launch file by one of them.

HappySamuel, 2019-06-24:
Hi @at-wat 

I have a question. If i use 3d lidar, then i need to use costmap_cspace/pointcloud2_to_map, while 2d lidar with costmap_cspace/laserscan_to_map. Right? If i have both 3d lidar and 2d lidar, can i remap both of their "map_local" to "overlay1"?

[image]

As only the "overlay1", which is type of Costmap3dLayerFootprint will affect the path planning, right?

Best,
Samuel

at-wat, 2019-07-06:
this demonstration setting has some useless layers.
```yaml
  static_layers:
    - name: unknown
      type: Costmap3dLayerUnknownHandle
      unknown_cost: 100  # Treat unknown cells in the global map as occupied
  layers:
    - name: overlay1
      type: Costmap3dLayerFootprint
      overlay_mode: max
      linear_expand: 0.15  # Expand obstacle by 0.15m
```
and feed local costmap to `/overlay1` should work.

> If i have both 3d lidar and 2d lidar, can i remap both of their "map_local" to "overlay1"?

It will cause local costmap chattering.
I think you should convert 2D LaserScan to PointCloud2 and remap two PointCloud2 topics (3D lidar cloud and 2D lidar cloud) to pointcloud2_to_map input in this case.

Or you can add another overlay layer like:
```yaml
  layers:
    - name: overlay_2d
      type: Costmap3dLayerFootprint
      overlay_mode: max
      linear_expand: 0.15  # Expand obstacle by 0.15m
    - name: overlay_3d
      type: Costmap3dLayerFootprint
      overlay_mode: max
      linear_expand: 0.15  # Expand obstacle by 0.15m
```
and remap pointcloud2_to_map output to `/overlay_3d` and laserscan_to_map output to `/overlay_2d` should also work. But this version consumes more CPU power to handle additional overlay layer.

HappySamuel, 2019-07-08:
Hi @at-wat 

For the 1st solution (convert 2D laserscan to pointcloud2 and remap two pointcloud2 topics to pointcloud2_to_map input) you mentioned, can 1 topic share two input topics?? Won't there is gonna to have conflicts?

For the 2nd solution i will try that out.

Best,
Samuel

at-wat, 2019-07-08:
> For the 1st solution (convert 2D laserscan to pointcloud2 and remap two pointcloud2 topics to pointcloud2_to_map input) you mentioned, can 1 topic share two input topics?? Won't there is gonna to have conflicts?

pointcloud2_to_map and laserscan_to_map internally have a buffer.
You can send pointclouds from multiple sources to them.

HappySamuel, 2019-07-08:
Hi @at-wat 

Ok, i will try them out. Thank you for the information and guidance.

Best,
Samuel

closed 2019-07-21 by HappySamuel