at-wat/neonavigation #331: Footprint and Pathvelocity don't show up on RVIZ - TF error

HappySamuel, 2019-06-21:
Hi

I change the dummy robot to my own robot model, but i couldn't have footprint and pathvelocity to be visualized on the RVIZ. Whenever i use the dummy robot, these two are shown properly. The error shows that is due to the TF error. The picture below showed the error message on RVIZ:
[image]

Here's the launch file:
[image]

Here's the tf tree:
[image]

Please advise what should i do to have them to be visualized on RVIZ. Thank you very much.

Best,
Samuel

at-wat, 2019-06-21:
It looks like not a neonavigation problem but a general simulation time configuration problem.
The error from rviz said that the data you want to show has a timestamp of current wall time, but the latest transform timestamp points 50 years ago, or 169 seconds after Unix time epoch.
I guess `use_sim_time` setting is incorrect.

HappySamuel, 2019-07-21:
Hi @at-wat 

Thank you for helping, this error is indeed raised due to ``` use_sim_time ``` setting is incorrect. I manage to get rid of this error.

Best,
Samuel

closed 2019-07-21 by HappySamuel