at-wat/neonavigation #341: Guidance Needed [Trajectory Server, Trajectory Recorder, Trajectory Saver, Trajectory Tracker]

HappySamuel, 2019-07-05:
Hi

I successfully recorded the trajectory with the trajectory_recorder. I also successfully saved the trajectory using the trajectory_saver. The trajectory i recorded is shown as below:
[image]

However when i wish to run this recorded path with trajectory_tracker. But the recorded path doesn't show up. And the trajectory_tracker didn't output the cmd_vel at all. Here i attached the screenshot.
[image]

My launch file for the trajecrtory_server and trajectory_tracker is as below.
[image]

Please advice what shall i do to get the trajectory_tracker to run the recorded path from trajectory_server. Thank you.

Best,
Samuel

at-wat, 2019-07-06:
trajectory_saver/server are not well maintained for long long time and it looks not publishing a path at node startup.
Could you try calling `/change_path` service specifying your path file?

BTW, it would be better to paste text based data like launch and yaml as a code block instead of putting as an image.

HappySamuel, 2019-07-08:
Hi @at-wat 

Yes, i tried the /change_path service, and it can work. the command i used is as following.

```
rosservice call /change_path "{header: {seq: 0, stamp: {secs: 0}, frame_id: ''}, filename: '/home/samuel/try-ws/src/atlas200mini-pkgs/atlas200mini_gazebo/traj/maze_traj_test_1.path', id: 0}"
```
Just curious, why do we need to send the frame_id, seq, id ? Isn't the filename is the only thing we need?

Best,
Samuel

at-wat, 2019-07-08:
OK, so trajectory_server is better to be fixed it to load a path at node startup, if filename is given as a parameter.
Unfortunately, I don't have time to do it soon at now. (If you can make a PR, it would be faster.)

> Just curious, why do we need to send the frame_id, seq, id ? Isn't the filename is the only thing we need?

hmm, I don't remember at all. It looks added when it was not yet opensource.

HappySamuel, 2019-07-08:
Hi @at-wat 

What is PR that you mentioned? 

Besides, I am thinking to make a node that can record the path with velocity, as i noticed that the trajectory_tracker tracks the /path_velocity, which is published by planner_3d. While the linear_velocity is denoted as "nan". Attached picture is the rostopic echo /path_velocity.
[image]

Just a quick question, will the trajectory_tracker try its best to maintain the speed of moving as receiving in the /path_velocity, if the speed is mentioned.

Best,
Samuel

at-wat, 2019-07-08:
> What is PR that you mentioned?

It's about trajectory_server's initial path loading.
It will take a while to be fixed by me due to my time. If someone could fix it by a PR, it's great.

> will the trajectory_tracker try its best to maintain the speed of moving as receiving in the /path_velocity, if the speed is mentioned.

Yes, if linear_velocity.x is not `nan`, trajectory_tracker tries to limit the velocity by this value.

HappySamuel, 2019-07-10:
Hi @at-wat 

Here's the code of trajectory_recorder that contains the ability of:
(1) Record Path with Speed
(2) Path recorded based on dist2d and ang2d
(3) Save the recorded path upon shutting down this node
```
#!/usr/bin/env python

"""
Author : Samuel Chieng Kien Ho
Reference : Atsushi Watanabe
https://github.com/at-wat/neonavigation/blob/master/trajectory_tracker/src/trajectory_recorder.cpp

Function: (1) Record Path with Velocity (pathWV)
          (2) Path recorded according to dist2d and ang2d
          (3) Save the recorded pathWV upon shutdown of this node
"""

import rospy
import numpy as np
import time, os
import pickle
import tf2_ros
import PyKDL

from nav_msgs.msg import Path, Odometry
from geometry_msgs.msg import Twist, PoseStamped, Vector3
from trajectory_tracker_msgs.msg import PoseStampedWithVelocity, PathWithVelocity

class TrajectoryRecorder(object):
    def __init__(self):
        # Define Adjustable Parameters
        self.filepath = os.path.join(rospy.get_param("~filepath", "/home/samuel/try-ws/src/atlas200mini-pkgs/atlas200mini_gazebo/traj/"), time.strftime("%Y-%m-%d-%H-%M-%S") + ".path")
        self.frame_robot = rospy.get_param("~frame_robot", "base_link")
        self.frame_global = rospy.get_param("~frame_global", "map")
        self.dist_interval = float(rospy.get_param("~dist_interval", "0.3"))
        self.ang_interval = float(rospy.get_param("~ang_interval", "1.0"))

        # Internal USE Variables - Modify with Consultation
        self.rate = rospy.Rate(50)
        self.tf2Buffer = tf2_ros.Buffer()
        self.listener = tf2_ros.TransformListener(self.tf2Buffer)
        self.x_spd = float("NaN") # Default Value "NaN" means "Don't Care"
        self.save_path_WV = []

        # Publisher
        self.pathWV_pub = rospy.Publisher("path_velocity", PathWithVelocity, queue_size=1)

        # Subscribers
        self.twist_sub = rospy.Subscriber("atlas200mini_velocity_controller/cmd_vel", Twist, self.speedCB, queue_size=1)
        self.odom_sub = rospy.Subscriber("wheel_odometry/odom", Odometry, self.speedCB, queue_size=1)

        # Main Loop
        self.loop()

        # Save the recorded trajectory on shutdown
        rospy.on_shutdown(self.saveTrajectory)

    # Speed Callback Function - If there's speed input, speed will be recorded.
    def speedCB(self, msg):
        if isinstance(msg, Twist):
            self.x_spd = msg.linear.x
        elif isinstance(msg, Odometry):
            self.x_spd = msg.twist.twist.linear.x

    # Main Loop
    def loop(self):
        pathWV = PathWithVelocity()
        pathWV.header.frame_id = self.frame_global
[... 94 lines cut ...]
        # Service Server
        self.changePath_service = rospy.Service("change_path", ChangePath, self.change)

    def change(self, req_path):
        loaded_pathWV = PathWithVelocity()
        loaded_pathWV = self.loadFile(req_path.filename)
        loaded_pathWV.header.stamp = rospy.Time.now()
        self.pathWV_pub.publish(loaded_pathWV)
        return True

    def loadFile(self, request_path):
        with open(request_path) as infile:
            data = pickle.load(infile)
            return data

if __name__=="__main__":
    rospy.init_node("trajectory_server")
    TrajectoryServer()
    rospy.spin()
```

Hope these will help the neonavigation repository.

Best,
Samuel

closed 2019-09-28 by HappySamuel