at-wat/neonavigation #346: How to enable trajectory_tracker to do reverse driving?

HappySamuel, 2019-07-09:
Hi

i tested the trajectory_tracker and found that its result(forward driving) is nice. I wonder if i want to use it to do reverse driving, for whatever path that sent to it. How shall i enable this? using service call will be good.

Best,
Samuel

at-wat, 2019-08-15:
Path message with poses facing opposite to the direction of the path makes robot backward running.

Or, do you want to make the robot running reversely on the recorded path?
If yes, trajectory_tracker doesn't provide it.
In this case, please make a node or a tool which creates a reverse-path message from given path message.

HappySamuel, 2019-08-15:
Hi @at-wat 

I recently get the robot to be able to run reversely on the recorded path. Neither changes are needed to be made onto the trajectory_recorder nor the trajectory_planner which generate path. The robot will follow the given path reversely. Only slight modification is needed to achieve this. Will share with you soon. 

Best,
Samuel

closed 2019-09-28 by HappySamuel