at-wat/neonavigation #438: Running in the middle of two constrained wall (Geofence)

HappySamuel, 2020-02-18:
Hi 

How shall i use the planner_cspace planner or costmap_cspace to make the robot to run in the middle of two constrained wall (Geofence)?

Note: Geofence is generated by using largemap_to_map

Best,
Samuel

at-wat, 2020-03-28:
Sorry for the delayed reply.
In the demo launch in neonavigation_launch, extra static_layer `overlay0`
https://github.com/at-wat/neonavigation/blob/39a87cd6d717d324a3a21fcf4d99ac099e3997b5/neonavigation_launch/config/navigate.yaml#L9-L13
and extra layer `overlay2` 
https://github.com/at-wat/neonavigation/blob/39a87cd6d717d324a3a21fcf4d99ac099e3997b5/neonavigation_launch/config/navigate.yaml#L17-L22
are specified.
You can publish a global geofence map (OccupancyGrid) to `/overlay0` topic, and/or a local one to `overlay2` topic.

HappySamuel, 2020-03-30:
Hi @at-wat 

Thanks for the reply. I have several questions as below:
(1) How will the overlay0 and overlay2 enable the robot to run in the middle of two constrained wall? Path generated by cspace_planner is in the middle of two constrained wall.
(2) How do the linear_expand and linear_spread work? I tried to adjust them, but i yet to understand them.

Best,
Samuel

at-wat, 2020-03-30:
> (1) How will the overlay0 and overlay2 enable the robot to run in the middle of two constrained wall? Path generated by cspace_planner is in the middle of two constrained wall.

I might misread your original question.
Do you mean that you want to avoid running the robot close to the geofence wall?
If so, it would be better to spread/expand the wall.

> (2) How do the linear_expand and linear_spread work? I tried to adjust them, but i yet to understand them.

`linear_expand` and `linear_spread` works like below:
```
input
0 0 0 0 0
0 0 0 0 0
0 0 9 0 0
0 0 0 0 0
0 0 0 0 0

linear_expand: 1grid
linear_spread: 0
0 0 0 0 0
0 0 9 0 0
0 9 9 9 0
0 0 9 0 0
0 0 0 0 0

linear_expand: 0
linear_spread: 2grids
0 1 3 1 0
1 4 6 4 1
3 6 9 6 3
1 4 6 4 1
0 1 3 1 0 (not a calculated result, but by hand)
```

HappySamuel, 2020-03-31:
Hi @at-wat 

Now i finally understand the concept of linear expand and linear spread. 
Thanks a lot for the explanation. 

Best,
Samuel

HappySamuel, 2020-08-31:
Hi @at-wat

The linear spread and linear expand, are they in grids or in meters?

Best,
Samuel

HappySamuel, 2020-09-07:
Hi @at-wat 

I am curious how we define the layers is global or local?
Are the global and local are differentiate via static layer and layers?

Best,
Samuel

at-wat, 2021-01-01:
> The linear spread and linear expand, are they in grids or in meters?

It's meters.

> I am curious how we define the layers is global or local?
> Are the global and local are differentiate via static layer and layers?

`static_layers`: all cells in the costmap are updated at once (usually for global map)
`layers`: the costmap can be partially (usually for local map)

> Besides, what's the ang_resolution?

costmap_3d and planner_3d calculate path in 3-DoF parameter space (x, y, and orientation). `ang_resolution` is a resolution of searching grid in orientation dimension.
So, denser ang_resolution results higher resolution of angle of resultant path (smoother) but computation cost increases for dense ang_resolution.

HappySamuel, 2021-01-06:
Hi @at-wat 

Thank you for the explanation.

Best,
Samuel

closed 2021-01-06 by HappySamuel