at-wat/neonavigation #482: planner_cspace.rosunit-test_debug_outputs/Hysteresis is flaky

at-wat, 2020-05-08:
```
build/test_results/planner_cspace/rosunit-test_debug_outputs.xml: 4 tests, 0 errors, 2 failures, 0 skipped
```
```
  /usr/bin/python2 /opt/ros/melodic/share/rostest/cmake/../../../bin/rostest --pkgdir=/catkin_ws/src/neonavigation/planner_cspace --package=planner_cspace --results-filename test_test_debug_outputs_rostest.xml --results-base-dir "/catkin_ws/build/test_results" /catkin_ws/src/neonavigation/planner_cspace/test/test/debug_outputs_rostest.test 
... logging to /root/.ros/log/rostest-a4a36d667cd1-5843.log
[ROSUNIT] Outputting test results to /catkin_ws/build/test_results/planner_cspace/rostest-test_test_debug_outputs_rostest.xml
[33m[ WARN] [1588826897.723299637]: overlay_mode of the static layer is not specified. Using MAX mode.[0m
[0m[ INFO] [1588826898.647774823]: Map received[0m
[0m[ INFO] [1588826898.652951555]:  linear_resolution 0.10 x (32x32) px[0m
[0m[ INFO] [1588826898.653110325]:  angular_resolution 0.39 x 16 px[0m
[0m[ INFO] [1588826898.654172610]:  origin 0.000 m, 0.000 m, 0.000 rad[0m
[0m[ INFO] [1588826898.839967835]: x:3, y:3 grids around the boundary is ignored on path search[0m
[0m[ INFO] [1588826900.295246342]: New goal received (9, 5, 8)[0m
99999999999999999999999999999999
99999999999999999999999999999999
99999999999999999999999999999999
99999999999999999999999999999999
99999999999999999999999999999999
99999999999999999999999999999999
99999999999999999999999999999999
99999999999999999999999999999999
99999999999999999999999999999999
99999999999999999999999999999999
99999999999999999999999999999999
99999999999999999999999999999999
99999999999999999999999999999999
99999999999999999999999999999999
99999999999999999999999999999999
99999999999999999999999999999999
99999999999999999999999999999999
99999999999999999999999999999999
99999999999999999999999999999999
99999999999999999999999999999999
99999999999999999999999999999999
99999999999999999999999999999999
99999999999999999999999999999999
99999999999999999999999999999999
99999999999999999999999999999999
99999999999999999999999999999999
99999999999999999999999999999999
99999999999999999999999999999999
99999999999999999999999999999999
99999999999999999999999999999999
99999999999999999999999999999999
99999999999999999999999999999999
path:
2.55, 0.55, 
2.5, 0.55, 
2.45, 0.55, 
2.4, 0.55, 
2.35, 0.55, 
2.3, 0.55, 
2.25, 0.55, 
2.2, 0.55, 
2.15, 0.55, 
2.1, 0.55, 
2.05, 0.55, 
2, 0.55, 
1.95, 0.55, 
1.9, 0.55, 
1.85, 0.55, 
1.8, 0.55, 
1.75, 0.55, 
1.7, 0.55, 
1.65, 0.55, 
1.6, 0.55, 
1.55, 0.55, 
1.5, 0.55, 
1.45, 0.55, 
1.4, 0.55, 
1.35, 0.55, 
1.3, 0.55, 
1.25, 0.55, 
1.2, 0.55, 
1.15, 0.55, 
1.1, 0.55, 
1.05, 0.55, 
1, 0.55, 
0.95, 0.55, 
[Testcase: testtest_debug_outputs] ... ok

[ROSTEST]-----------------------------------------------------------------------

[planner_cspace.rosunit-test_debug_outputs/Hysteresis][FAILURE]-----------------
/catkin_ws/src/neonavigation/planner_cspace/test/src/test_debug_outputs.cpp:175
      Expected: data.value
      Which is: '\0'
To be equal to: map_hysteresis_->data[addr]
      Which is: 'd' (100, 0x64)
x: 25, y: 4
--------------------------------------------------------------------------------

[planner_cspace.rosunit-test_debug_outputs/Hysteresis][FAILURE]-----------------
/catkin_ws/src/neonavigation/planner_cspace/test/src/test_debug_outputs.cpp:175
      Expected: data.value
      Which is: '\0'
To be equal to: map_hysteresis_->data[addr]
      Which is: 'd' (100, 0x64)
x: 20, y: 4
--------------------------------------------------------------------------------

[planner_cspace.rosunit-test_debug_outputs/Hysteresis][FAILURE]-----------------
/catkin_ws/src/neonavigation/planner_cspace/test/src/test_debug_outputs.cpp:175
      Expected: data.value
      Which is: '\0'
To be equal to: map_hysteresis_->data[addr]
      Which is: 'd' (100, 0x64)
x: 15, y: 4
--------------------------------------------------------------------------------

[planner_cspace.rosunit-test_debug_outputs/Hysteresis][FAILURE]-----------------
/catkin_ws/src/neonavigation/planner_cspace/test/src/test_debug_outputs.cpp:175
      Expected: data.value
      Which is: '\0'
To be equal to: map_hysteresis_->data[addr]
      Which is: 'd' (100, 0x64)
x: 10, y: 4
--------------------------------------------------------------------------------

[planner_cspace.rosunit-test_debug_outputs/Remembered][passed]

SUMMARY
[1;31m * RESULT: FAIL[0m
 * TESTS: 2
 * ERRORS: 0
[1;31m * FAILURES: 1[0m
```

Taka-Kazu, 2020-05-08:
As far as I've tried, 
```
$ catkin_make -DCMAKE_BUILD_TYPE=Release && catkin_make run_tests_planner_cspace && catkin_test_results
```
 passes the test, but 
```
$ catkin_make && catkin_make run_tests_planner_cspace && catkin_test_results
```
fails.

at-wat, 2020-05-14:
I finally found it is a bug due to uninitialized variable.
https://github.com/at-wat/neonavigation/blob/f2fbff68f43219c2bcb1c05da2d1b531d7bf39dc/planner_cspace/src/planner_3d.cpp#L1721

PR #486 (2020-05-14): planner_cspace: fix uninitialized variable
fix #482
closed 2020-05-14 by at-wat