at-wat/neonavigation #507: Time for path found too long

HappySamuel, 2020-06-17:
Hi

I am facing issue with the path found taking too long time. This cause the robot to move, stop, move, stop, move until reaching the goal. Below's the paramete
 that i used:

HappySamuel, 2020-06-19:
Hi @at-wat 

I tried to send the robot to go for the 2nd goal once it reached the 1st goal, and i found that the path found took longer time than going towards the 1st goal. When you try with 3rd goal, the time consumed for path found become even longer. 
Eg:
Going to 1st goal (path found is taking 2-3 secs when each local path is generated)
Going to 2nd goal (path found is taking 5-6 secs when each local path is generated)
Going to 3rd goal (path found is taking 9-10 secs when each local path is generated)

As when the path found took longer, the robot already arrived the end of the local path and it will stop and wait for the next local path. This cause the robot to behave move, stop, move, stop, move, stop, move.

Do you know why it's happening like this? Is there a solution for this issue? 

Thanks.

at-wat, 2020-06-19:
I have never seen such behavior. In my use case, the path planner works with at least a few Hz.
You can see the detailed computation duration of planning steps by setting log verbosity level to debug.

HappySamuel, 2020-06-19:
Hi @at-wat 

Yes, i have enable the `print planning duration` . Above eg. are the result it showed to me. The time taken for path planning become longer and longer after receiving new goal after completed the current one. (Not the temporary goal in local path it created, it's the goal sent via move_base action).

HappySamuel, 2020-11-18:
Hi @at-wat 

I have tested your latest code in the gazebo simulation, the issue of `time for path found too long` has been ruled out. I found that it's due to i am using outdated `planner_3d` (0.4.0) code. Now, the path generated is fast. Your work is good.

Best,
Samuel

closed 2020-11-18 by HappySamuel