at-wat/neonavigation #510: neonavigation with noetic

altineller, 2020-07-01:
Hello,

I managed to catkin_make neonavigation stack with ubuntu:20.04 noetic, but I had no luck. It will not give any errors, but the rviz screen does not look anything like on melodic, and it seems it will not publish maps.

Any ideas/recommendations or pointers in the right direction are greatly appreciated.

Best regards,
C.

at-wat, 2020-07-16:
> the rviz screen does not look anything like on melodic, and it seems it will not publish maps.

In the neonavigation's demo launch file, map is loaded by usual `map_server`. So, if map is not shown on your rviz screen, it should be an environment specific problem of rviz, map_server or ros core library.

I just tested the latest release of neonavigation (installed via apt) on docker environment, and the demo works.
[image]

at-wat, 2021-01-01:
Seems not a problem in neonavigation package. Feel free to reopen or open a new issue if you have any other questions.

closed 2021-01-01 by at-wat