at-wat/neonavigation #550: Guidance needed for implementing Highway into Costmap

HappySamuel, 2020-08-31:
Hi @at-wat 

I would like to implement the highway concept into your costmap planner, learn it from the Nvidia isaac SDK.
Step 1: Convert a drawn the line of 2d map into highway for costmap
Step 2: Global Path Planner will prioritize the highway and plan the path to use highway.
Step 3: When it's on the highway, the local planner will generate the path that lie on the highway, until there's a need to exit highway to get to its location.

Any guidance on how shall i do/modify to achieve these 3 steps.

Best
Samuel

at-wat, 2021-01-01:
I assume `empty cell/grid` means unknown cell (-1 in OccupancyGrid).

> (1) What's the default cost for each empty cell/grid?
> (2) How to define the default cost for the empty cell/grid from the global map?

Costmap layers like `footprint` inherits cost of unknown cell from base layer.
`unknown_handle` layer applies cost of unknown cell by the parameter of `unknown_cost`. This can set default cost in costmap_3d layer.

If `unknown_handle` layer is not used, costmap_3d node outputs unknown cell as unknown.

planner_3d node has a parameter `unknown_cost` to set cost value to unknown cells. (default: 100 = occupied)

HappySamuel, 2021-01-06:
Hi @at-wat 

Thank you for the explanation. Recently, i successfully implement the highway concept using the navigation stack. But still yet to implement with your costmap_3d. Will share with you if i successfully implement the highway into your costmap_3d.

Best,
Samuel