at-wat/neonavigation #565: Visualize Global Path?

HappySamuel, 2020-11-18:
Hi

Is there a way to visualize the global path from robot pose to target goal, like the global path in navigation stack? So far i only found the local path that generated.

Best,
Samuel

at-wat, 2021-01-01:
`planner_3d` doesn't have a single path as global path from the start to the goal.
But it calculates all paths to the goal from all possible positions as a distance map which is published to `/planner_3d/debug` when `debug_mode` parameter is `"cost_estim"`.

[image]
Gradation in this snapshot is a distance_map in planner_cspace and the global path from any start is determined by following the most steepest gradient direction from the start.

HappySamuel, 2021-01-06:
Hi @at-wat 

Thanks for the explanation. I found the path generated via planner_3d is very feasible for the robot to track, better than navigation stack local planner. If i want to make use of your planner_3d to generate a local path from the navigation stack global planner, can help to advise a direction or steps to be done to achieve this?

Best,
Samuel