at-wat/neonavigation #632: Goal not accepted with real robot

FrGe2016, 2021-12-18:
Hi great package

The robot is working perfectly when i use the normal navigation stack

I have modified the neonavigation demo.launch file to test it with my linorobot and nothing is planned. Here are the info. 

**[ERROR] [1639844616.742719628]: Goal [map] pose must be in the map frame [].**

But the goal pose in in the pap frame

**rostopic echo /move_base_simple/goal**
header: 
  seq: 0
  stamp: 
    secs: 1639843216
    nsecs: 753899602
  frame_id: **"map"**
pose: 
  position: 
    x: 8.681979179382324
    y: 3.8665976524353027
    z: 0.0
  orientation: 
    x: 0.0
    y: 0.0
    z: -0.9830096719102385
    w: 0.18355376577702048
---

My launch file for my physical linorobot is :

<?xml version="1.0"?>
<launch>
  <arg name="use_path_with_velocity" default="true" />
  <arg name="vel" default="0.5" />

  <include file="$(find neonavigation_launch)/launch/navigate.launch">
    <arg name="simulate" value="**false**" />
    <arg name="map_file" value="$(**find linorobot)/maps/house.yaml**" />
    <arg name="use_path_with_velocity" value="true" />
    <arg name="vel" value="$(arg vel)" />
  </include>

  **
  <include file="$(find linorobot)/launch/include/amcl.launch" />**

  <node pkg="map_server" type="map_server" name="map_server_local" args="$(**find linorobot)/maps/house.yaml**">
    <remap from="map" to="map_with_local_objects" />
  </node>
  <node pkg="costmap_cspace" type="largemap_to_map" name="largemap_to_map">
    <remap from="map" to="map_with_local_objects" />
    <remap from="map_local" to="overlay1" />
    <param name="width" value="60" />
    <param name="hz" value="1.5" />
    <param name="round_local_map" value="true" />
    <param name="simulate_occlusion" value="true" />
  </node>

  <node pkg="rviz" type="rviz" name="rviz" args="-d $(find neonavigation_launch)/config/visualization.rviz" respawn="true" />
</launch>

at-wat, 2021-12-18:
https://github.com/at-wat/neonavigation/blob/e17cf72d4f980ce33f8e5e22c52e3180015e4f65/planner_cspace/src/planner_3d.cpp#L390-L391

> `[ERROR] [1639844616.742719628]: Goal [map] pose must be in the map frame [].`

It means `planner_3d` hadn't received map yet or had received map with empty `frame_id`.

FrGe2016, 2021-12-19:
But the map is shown in rviz ?

FrGe2016, 2021-12-19:
Finally i solved it by making changes in the navigate.launch file

1) I have moved the launching of the map_server node before the costmap_3d node. ( this is the main reason why the goal was not in the map frame ) 

2) I have removed if="$(arg simulate)" in the setting of the footprint parameter.

closed 2021-12-19 by FrGe2016