at-wat/neonavigation #655: Flaky tests

Taka-Kazu, 2022-12-28:
# trajectory_tracker
- https://github.com/at-wat/neonavigation/actions/runs/3791339351/jobs/6446850442
```
2022-12-28T03:29:49.4365174Z Scanning dependencies of target _run_tests_trajectory_tracker_rostest_trajectory_tracker_rostest__odom_delay_0.04__use_odom_true__use_time_optimal_control_true.test
2022-12-28T03:30:35.0212928Z terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::bad_weak_ptr> >'
2022-12-28T03:30:35.0213398Z   what():  tr1::bad_weak_ptr
2022-12-28T03:30:35.6758256Z ... logging to /root/.ros/log/rostest-2d6cac69262b-8892.log
2022-12-28T03:30:35.6759604Z [ROSUNIT] Outputting test results to /catkin_ws/build/test_results/trajectory_tracker/rostest-trajectory_tracker_rostest__odom_delay_0.xml
2022-12-28T03:30:35.6760305Z [Testcase: testtest_trajectory_tracker] ... FAILURE!
2022-12-28T03:30:35.6760659Z FAILURE: test [test_trajectory_tracker] did not generate test results
2022-12-28T03:30:35.6761037Z   File "/usr/lib/python2.7/unittest/case.py", line 329, in run
2022-12-28T03:30:35.6761305Z     testMethod()
2022-12-28T03:30:35.6761726Z   File "/opt/ros/melodic/lib/python2.7/dist-packages/rostest/runner.py", line 164, in fn
2022-12-28T03:30:35.6762153Z     self.assert_(os.path.isfile(test_file), "test [%s] did not generate test results"%test_name)
2022-12-28T03:30:35.6762524Z   File "/usr/lib/python2.7/unittest/case.py", line 422, in assertTrue
2022-12-28T03:30:35.6762830Z     raise self.failureException(msg)
2022-12-28T03:30:35.6763260Z --------------------------------------------------------------------------------
2022-12-28T03:30:35.6763483Z 
2022-12-28T03:30:35.6763742Z [ROSTEST]-----------------------------------------------------------------------
2022-12-28T03:30:35.6763959Z 
2022-12-28T03:30:35.6764075Z [testtest_trajectory_tracker][failed]
```

# planner_cspace
- https://github.com/at-wat/neonavigation/actions/runs/3791339351/jobs/6447022614
```
2022-12-28T03:49:33.9643317Z Scanning dependencies of target _run_tests_planner_cspace_rostest_test_test_preempt_rostest.test
2022-12-28T03:49:34.8937908Z [33m[ WARN] [1672199374.891240577]: TF exception: "base_link" passed to lookupTransform argument target_frame does not exist. [0m
2022-12-28T03:49:34.9176717Z [33m[ WARN] [1672199374.916323041]: TF exception: "base_link" passed to lookupTransform argument target_frame does not exist. [0m
2022-12-28T03:49:34.9637373Z [33m[ WARN] [1672199374.954514490]: TF exception: "base_link" passed to lookupTransform argument target_frame does not exist. [0m
2022-12-28T03:49:34.9837662Z [33m[ WARN] [1672199374.982980076]: TF exception: "base_link" passed to lookupTransform argument target_frame does not exist. [0m
2022-12-28T03:49:35.0168325Z [33m[ WARN] [1672199375.016332494]: TF exception: "base_link" passed to lookupTransform argument target_frame does not exist. [0m
2022-12-28T03:49:35.1456302Z [31m[ERROR] [1672199375.145102935]: Goal [map] pose must be in the map frame [].[0m
2022-12-28T03:49:41.8360927Z ... logging to /root/.ros/log/rostest-a9bbb18e5a04-6416.log
2022-12-28T03:49:41.8361628Z [ROSUNIT] Outputting test results to /catkin_ws/build/test_results/planner_cspace/rostest-test_test_preempt_rostest.xml
2022-12-28T03:49:41.8362091Z [Testcase: testtest_preempt] ... ok
2022-12-28T03:49:41.8362277Z 
2022-12-28T03:49:41.8362588Z [ROSTEST]-----------------------------------------------------------------------
2022-12-28T03:49:41.8362846Z 
2022-12-28T03:49:41.8363165Z [planner_cspace.rosunit-test_preempt/Preempt][FAILURE]--------------------------
2022-12-28T03:49:41.8363609Z /catkin_ws/src/self/planner_cspace/test/src/test_preempt.cpp:107
2022-12-28T03:49:41.8364082Z Expected: (ros::Time::now()) < (deadline), actual: 1672199380.139205563 vs 1672199380.138642093
2022-12-28T03:49:41.8364538Z Action didn't get active: ABORTED(status: 0, error: 0)
2022-12-28T03:49:41.8365036Z --------------------------------------------------------------------------------
2022-12-28T03:49:41.8365291Z 
2022-12-28T03:49:41.8365297Z 
2022-12-28T03:49:41.8365389Z SUMMARY
2022-12-28T03:49:41.8365673Z [1;31m * RESULT: FAIL[0m
```

at-wat, 2022-12-28:
https://github.com/at-wat/neonavigation/actions/runs/3792072734/jobs/6447958477#step:10:10834
```
[ROSTEST]-----------------------------------------------------------------------

[safety_limiter.rosunit-test_safety_limiter2/SafetyLimitLinearSimpleSimulationWithMargin][FAILURE]
/catkin_ws/src/self/safety_limiter/test/src/test_safety_limiter2.cpp:95
Expected: (-0.90) > (x), actual: -0.9 vs -0.895511
--------------------------------------------------------------------------------

SUMMARY
 * RESULT: FAIL
 * TESTS: 1
 * ERRORS: 0
 * FAILURES: 1
 ```

at-wat, 2023-02-01:
https://s3-ap-northeast-1.amazonaws.com/alpine-ros-experimental.dev-sq.work/build_logs/5213e1ec-f566-4d38-b6ad-08d6e590875d

```
[ROSTEST]-----------------------------------------------------------------------

[safety_limiter.rosunit-test_safety_limiter2/SafetyLimitLinearSimpleSimulationWithMargin][FAILURE]
/home/builder/aports/ros/noetic/ros-noetic-safety-limiter/abuild/src/safety_limiter/test/src/test_safety_limiter2.cpp:90
Expected: (0.90) < (x), actual: 0.9 vs 0.895511
--------------------------------------------------------------------------------

SUMMARY
�[1;31m * RESULT: FAIL�[0m
 * TESTS: 1
 * ERRORS: 0
�[1;31m * FAILURES: 1�[0m
```

at-wat, 2023-02-24:
```
[ROSTEST]-----------------------------------------------------------------------

[trajectory_tracker.rosunit-test_trajectory_tracker/StraightStop][FAILURE]------
/catkin_ws/src/self/trajectory_tracker/test/src/test_trajectory_tracker.cpp:73
The difference between pos_[0] and 0.5 is 0.026499639653017382, which exceeds error_lin_, where
pos_[0] evaluates to 0.52649963965301738,
0.5 evaluates to 0.5, and
error_lin_ evaluates to 0.025000000000000001.
[overshoot after goal (0)]
--------------------------------------------------------------------------------

[trajectory_tracker.rosunit-test_trajectory_tracker/StraightStopOvershoot][passed]
[trajectory_tracker.rosunit-test_trajectory_tracker/StraightStopConvergence][passed]
[trajectory_tracker.rosunit-test_trajectory_tracker/StraightVelocityChange][passed]
[trajectory_tracker.rosunit-test_trajectory_tracker/CurveFollow][passed]
[trajectory_tracker.rosunit-test_trajectory_tracker/InPlaceTurn][passed]
[trajectory_tracker.rosunit-test_trajectory_tracker/SwitchBack][passed]
[trajectory_tracker.rosunit-test_trajectory_tracker/SwitchBackWithPathUpdate][passed]

SUMMARY
 * RESULT: FAIL
 * TESTS: 8
 * ERRORS: 0
 * FAILURES: 1
```

at-wat, 2023-03-07:
done

closed 2023-03-07 by at-wat